supermerill/SuperSlicer #3003: App crashes switching to volumetric flow

Fission360, 2022-07-29:
### What happened?

SS app crashes when switching to "volumetric flow rate" under the gcode preview "view" boxes in the bottom left corner.

### Project file & How to reproduce

[Fan Mount-Voron V0.1-PLA (PLAID).zip](https://github.com/supermerill/SuperSlicer/files/9219021/Fan.Mount-Voron.V0.1-PLA.PLAID.zip)

### Version

2.3.57.12

### Operating system

Windows 11 x64

### Printer model

Voron 0.1

supermerill, 2022-07-30:
As you link a gcode file, i guess the problem is with the gcode viewer?
I launched it but changing the view to "volumetric flow rate" didn't crash it.
I tried on 2.3.57.12 and my dev version.

But I can see that the gcode-viewer is completely buggy with this gcode.

Fission360, 2022-07-30:
Yeah, it only seems to happen after I slice the file and switch. For now I just swapped to one version older.