supermerill/SuperSlicer #3004: Arachne

drveni, 2022-07-30:
### What happened?

Feature request please... im put this topic in Feature but they close and and told me to open it here.

### Project file & How to reproduce

in the new prusaslicer alpha version it has a mod called Arachne https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha3 . helps in the quality of printing. please watch the video with the attached link for an explanation so I don't write. 

https://toms3d.org/2022/07/28/arachne-in-prusaslicer-0-4mm-nozzles-just-became-obsolete/ link  for a detailed explanation of what exactly it does

it would be nice if you implemented it.

### Version

2.4.58.3

### Operating system

M1 mac

### Printer model

ender

mojamuto, 2022-07-30:
just wait for superslicer to merge with prusaslicer 2.5

Pavulon87, 2022-08-01:
From July 29, arachne is available on the [actions](https://github.com/supermerill/SuperSlicer/actions) page ([nightly_dev](https://github.com/supermerill/SuperSlicer/commits/nightly_dev) branch)

mojamuto, 2022-08-02:
> From July 29, arachne is available on the [actions](https://github.com/supermerill/SuperSlicer/actions) page ([nightly_dev](https://github.com/supermerill/SuperSlicer/commits/nightly_dev) branch)

if I am not mistaken, the apple m1 is arm, and the dev compilation of arm macos failed.
On linux it failed and I have not been able to test that version yet.

supermerill, 2022-08-04:
Arm can't be built anymore in the mac intel worker (or i don't know how)
I have to build it manually. I won't do that for nightly.

megapro17, 2022-10-05:
and how to enable it after downloading nightly_dev build? there's no perimeter generator in settings

Stonedge, 2022-10-13:
> and how to enable it after downloading nightly_dev build? there's no perimeter generator in settings
clearing the ui_layout folder did the trick for me... hope it help
https://github.com/supermerill/SuperSlicer/issues/3293

closed 2024-02-11 by supermerill