supermerill/SuperSlicer #3005: ERROR at FreePySCAD

Hashan0, 2022-07-30:
### Version
2.5.59.0  (Nightly windows x64 #597)

### Operating system type + version
Windows 10

### Behavior

ERROR at FreePySCAD 

* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[image]
_

supermerill, 2022-08-04:
Can you make a screenshot with the slicer in english version?

Hashan0, 2022-08-04:
Version 2.4.58.3 OK.

Version 2.5.59.0 crash due to error
[image]

Have a test of FreeCAD-Path from "C:\....\FreeCAD 0.20" to nonexistent directory "C:\....\FreeCAD 0.21"  changed.

[image]

Error message Version 2.4.58.3
[image]

Error message Version 2.5.59.0
[image]

supermerill, 2022-08-04:
Dang, right. I forgot to remove the linux test.

closed 2024-02-11 by supermerill