supermerill/SuperSlicer #3013: When support is disabled, no skirt gets generated

Zamundaaa, 2022-08-01:
This is a recent change; I think this happened when I updated from 2.4.58.2. I'm not sure if it's intentional or not, but it's really unintuitive and it took me a while to find out why the skirt wasn't there

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: 2.4.58.3

Zamundaaa, 2022-08-01:
hmm, it happens for me with all profiles and objects. Here you go: [Calibration cube without support.zip](https://github.com/supermerill/SuperSlicer/files/9234465/Calibration.cube.without.support.zip)

Pavulon87, 2022-08-01:
SuperSlicer_2.4.58.3_win64_220714:
[image]
Maybe it is OS dependent bug