supermerill/SuperSlicer #3018: How do I get rid of the comments that this program generates?

Jeredadams, 2022-08-02:
2.4.58.3 Version
_Version of SuperSlicer used goes here (help->about)_

Win 10 64 Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
I am trying to get rid of all the lines that have been commented out. I don't like all that bullshit. I am a user that is coming from Simplify 3D and there method is a very simple script that gets entered into the program its self. Can I do this with this program? Also, my tool change script is added straight to the box for gcode tool changes. Is this possible with SS? I have attached a screen shot from Simplfy showing both the "Strip" command I am using, and the tool change code I have. 

Please let me know if this is possible in SuperSlicer. 

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[image]
_

supermerill, 2022-08-09:
> Also, my tool change script is added straight to the box for gcode tool changes

The csutom gcode for toolchange? write yoru macro inside.

> I am trying to get rid of all the lines that have been commented out. 

All the comment? 
You can remove most of them with a g-code substitution (regexp "`;.*\n`"  -> "")
there is still the header, that is used by many printing servers, and the config at the bottom, that is used to display the gcode in the gcode-viewer.

Jeredadams, 2022-08-10:
Thanks for getting back to me.

I am not sure how to use that or where to put it so that there are no
comments in my code.

I have attached a copy of what my code looks like for a reference.

How do I get it to look like mine? I don't like all the extra crap being
put in the file.

*Jered Adams *

LazeMSS, 2023-01-01:
@supermerill could it be possible for the g-code substion to allow to change the entire file and not a post process script - that way we can hack/make the Anker user thumbnail - it will only show thumbnails if the slicer name in the gcode file is "Generated with AnkerSlicer"