supermerill/SuperSlicer #3019: Feature request - Z-hop and travel at the same time.

simpat1zq, 2022-08-02:
So I have tall prints that needs z-hop because sometimes a corner will curl up or something and the print would get knocked over without zhop. 

However, because of zhop, it leaves tiny wisps of stringing. If I disable z-hop, there is no stringing at all. I think is is because the printer pauses, goes straight up, and then travels. 

I would love for there to be a feature that does the z travel at the same time as the travel. There would of course need to be some considerations. The vertical move would need to start just after it leaves the print, and it would need to be at max height before it crosses anything on th
[image]
e layer or 2 below the current layer. 

Not only would something like this create cleaner prints, it should shave off a few minutes on prints using z-hop. The second or so it takes to do the zhop would pretty much be cut down to 0.  

I've attached a picture of what I'm trying to say here.

neophyl, 2022-08-02:
Sounds a lot like https://github.com/supermerill/SuperSlicer/issues/927

simpat1zq, 2022-08-02:
You're right. I'll close this one and move some of this info into that one. I searched for z-hop and it didn't pick up that one in the search.

closed 2022-08-02 by simpat1zq