supermerill/SuperSlicer #3020: Add checkbox to toggle interior brim

HitLuca, 2022-08-03:
I tried to look for it but I didn't find any mentions of it:
Could we get a toggle for the interior brim, in order to easily switch it off if we need to? It would be similar to the brim checkbox we already have near the infill percentage in the main window

supermerill, 2022-08-04:
You have to create a new scripted widget.

* help -> show configuration folder
* go to ui_layout/default and open the freq_fff.ui and print.as
* in freq_fff.ui, add `setting:simple:script:bool:depends$brim_width_interior:label$Interior Brim:tooltip$Set the interior brim. Will be set to 5mm if nothing was previously set.:s_brim_interior` above the `freq_purging_volumes` line.
* in print.as, paste this somwhere, like at the end of the file or a below the d_brim_set function. (it's the s_brim fucntion but with _interior added)
```

float last_brim_interior_val = 5;

int s_brim_interior_get()
{
	float bw = get_float("brim_width_interior");
	if (bw > 0) {
		last_brim_interior_val = bw;
		return 1;
	}
	return 0;
}

void s_brim_interior_set(bool new_val)
{
	if(new_val) {
		float bw = get_float("brim_width_interior");
		set_float("brim_width_interior", last_brim_interior_val);
	} else {
		set_float("brim_width_interior", 0);
	}
}
``` 
* now restart the slicer to verify it works.
* to be sure it won't be erased by a new veriosn, copy-paste the "default" folder, rename it, change the name & description in version.ini
* in the slicer, after a restart, go to preference ->GUI->Settings layout and colors, and select your new layout.

HitLuca, 2022-08-04:
Couldn't have asked for a better answer!

closed 2022-08-04 by HitLuca

HitLuca, 2022-08-30:
The new SuperSlicer update breaks the above code. This new version fixes it, assuming your `brim_width_interior` is set to 0 in order to match the exterior brim size:

in freq.fff update the entry with

```txt
setting:simple:script:bool:depends$brim_inside_holes:label$Interior Brim:tooltip$Set the interior brim set:s_brim_inside_holes		
```

in `print.as`, update the interior brim functions with

```txt
int s_brim_inside_holes_get()
{
	if (get_bool("brim_inside_holes")) {
		return 1;
	}
	return 0;
}

void s_brim_inside_holes_set(bool new_val)
{
	set_bool("brim_inside_holes", new_val);
}
```

personally I like the new brim addition, since most of the time we don't care about different brim sizes, this approach works for me!

supermerill, 2022-08-30:
I'm sot sure here.

If you have a brim width but no "interior brim", then you can select the "brim inside holes" that allow to put brim over object inside objects.

If you have "interior brim", I guess it's automatically activated? I may have to document that a bit more, I don't remember.