supermerill/SuperSlicer #3021: Cura-style dwell for short layers

mmirate, 2022-08-03:
**Is your feature request related to a problem? Please describe.**

Models whose topmost layers have intricate details and reduced cross-section, rely on layer-time minimums in order to maintain structural integrity of those details. Like PrusaSlicer, SuperSlicer in turn uses cooling and uniform slowdowns in order to produce a minimum layer time.

However, for sufficiently small top-layers of models who otherwise fill up the bed and thus cannot be printed in duplicate, uniform slowdowns are insufficient. And for various reasons (PC fan too small, hotend sock is broken and awaiting replacement, etc etc), the machine might be unable to provide sufficient partcooling to sufficiently compensate.

**Describe the solution you'd like**
Cura's minimum layer time does not (by default?) use slowdowns nor part-cooling, but rather simply dwells at the end of the layer in order to compensate for the layer time being otherwise below the minimum. I'd like that as a third option for layer-time mechanisms in SuperSlicer.

**Describe how it would work**
Below the two existing "short layer times" sections of the Cooling section in the Filament settings, add a third section which takes a minimum layer time, which should be less than the other two sections' minimum layer times. For any layer, if the layer time after applying slowdowns is still lesser than this minimum due to "Max speed reduction" and/or "Min print speed"; then at the end of the layer, retract, park somewhere nearby, dwell for a particular duration, then travel back and deretract for the next layer; and set the "particular duration" such that the resulting layer time is now equal to the minimum layer-time specified in the new config section.

**Describe alternatives you've considered**
I don't know of any alternatives to this, given a model with important features at the top whose middle layers occupy more than half the bed's XY space, and a machine with a broken sock or a weak part-cooling fan.

**Additional context**
None.

supermerill, 2022-08-04:
> alternatives 

If you know the layers, you can add a kind of pause in a conditional gcode in the layer change
```
{if layer_z > 20.5}
G1 Z{layer_z+1} F10
G1 Z{layer_z} F1000
{endif}
```
I don't know if it's effective or not

Pavulon87, 2022-08-04:
Perhaps it would be good to add `G4 P<milliseconds>` as well.

supermerill, 2022-08-04:
oh, yes, G4 is better than G1. I just couldn't find that gcode in the list. I search for "pause" and it's "dwell".