supermerill/SuperSlicer #3026: Artifacts on walls

ruredi87, 2022-08-04:
### What happened?

I see artifacts on outer walls. Prusa does not have when using classic perimeter generator but it has a bit with arachne.

### Project file & How to reproduce

[image]
[sides (2).zip](https://github.com/supermerill/SuperSlicer/files/9260824/sides.2.zip)

### Version

latest mosaic version but also has in nightly

### Operating system

windows 11

### Printer model

ender 3 v2

royal32, 2022-08-04:
I also noticed this on the nightly. Loaded the same 3mf into Prusaslicer 2.5.0 Alpha 3 and saw no artifacts. This might be of interest- on this model, when i reduce "Add middle perimeter threshold" to 54% from the default 75%, the artifacts disappear. At 55% they re-appear. 

[image]
[image]
[ArachneArtifacts.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9261118/ArachneArtifacts.3mf.zip)

ruredi87, 2022-08-04:
I found the reason of issue. Mine slice resolution was 0.01mm, after I turned to its default problem is solved but I dont know why its happened.

supermerill, 2022-08-04:
I think you want to deactivate/lower the `model rounding precision`. Thsi setting remove points if they are almost aligned. As your object has a very shallow curve, it trigger this failguard, and keep different points for different layers.

royal32, 2022-08-04:
Changing model rounding precision doesn't affect this issue for me- I tried 0, 0.000001, 0.0001, and 0.1. No change in the artifacts.

supermerill, 2022-08-04:
@royal32 I don't have your project, so i can' do anything.

royal32, 2022-08-04:
Huh... i dragged the zip into the first message but it didn't actually attach the file, I'll try again.
[ArachneArtifacts.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9263835/ArachneArtifacts.3mf.zip)

supermerill, 2022-08-05:
I removed some bugs from the archane (mostly malformed extrusion). But your problem is still here.
When prusa will move out of beta, i'll come back to check if it's still here.