supermerill/SuperSlicer #3027: bridge infill interrupted by overhang perimeters resulting in mid air print

rolandjohann, 2022-08-04:
### What happened?

As seen in the pictures I want to print a cylinder with a cutout cylinder from the inner bottom that gets smaller after some layers. To achieve that without supports, I inserted an artificial layer acting as bridge where the small diameter cylinder cutout perimeters can be printed on. Unfortunately the slicer doesn't treat the bridge as bridge only, but interrupts it with overhang perimeters.

Can someone explain what's happening here and if it's solvable by tweaking settings, or if it's a bug?

[image]

### layer 48
[image]

### layer 49
[image]

### layer 50
[image]

### Project file & How to reproduce

[adapter_problem.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9261615/adapter_problem.3mf.zip)
Slice and navigate to layer 49

### Version

2.4.58.3

### Operating system

macos 12.1

### Printer model

RatRig V-Core 3

Pavulon87, 2022-08-05:
Maybe try to reduce layer height or increase artificial layer height, or change `No perimeters on bridge areas` to `Fill the voids with bridges`.

49:
[image]

50:
[image]

rolandjohann, 2022-08-05:
@Pavulon87 thanks a lot. That resolves it.

Can you explain why superslicer behaves like this and we need to take care about bridges with perimeters manually? I want to understand the slicer better.

supermerill, 2022-08-05:
You have "only one perimeter on top" activated.
This setting works by detecting areas with top surface and excluding them from the perimeter generation.
Here your sacrificial layer is a single layer, and so contain a top surface. Perimeters will be created where the top surface ends + 'default infill anchor'.

* If you increase the default anchor (to 350%) , the perimeters are so pushed that they go outside of the bridge and disappear.
* Or if you remove the 'only one perimeter on top' (for all layer or only this one with a height range modifier), the problem disappear.
* Or if you use the 'No perimeters on bridge area -> fill the voids' like said on the answer above, then it will create a sacrificial layer even if the model doesn't have one! (you can also apply it only at the layer you want with a height range modifier)

rolandjohann, 2022-08-07:
@supermerill thanks for the explanation. The problem is now clear to me.

closed 2022-08-07 by rolandjohann