supermerill/SuperSlicer #3031: Wipe tower brim width incorrectly calculated

Pavulon87, 2022-08-05:
### What happened?

Previous versions of SuSi had the default value of 150% and everything was fine(now it is 2mm).
But in latest build something is wrong in calculation algorithm and brim could be bigger than a build plate when using percentage value.

150%
[image]

15%
[image]

### Project file & How to reproduce

[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/9265674/Shape-Box.zip)

### Version

2.4.58.3 / 2.5.59.0

### Operating system

windows 10

### Printer model

not relevant

Pavulon87, 2022-08-05:
I'm not sure, but maybe the point is that percentages are used as mm?
```
m_wipe_tower_brim_width(float(config.wipe_tower_brim_width)),
(...)
size_t loops_num = (m_wipe_tower_brim_width + spacing / 2) / spacing;
```

Previously:
`double unscaled_brim_width = m_config->wipe_tower_brim.get_abs_value(m_nozzle_diameter);`

supermerill, 2022-08-05:
Correct. Maybe another merge quirk.
Merged for next nightly

closed 2022-08-10 by Pavulon87