supermerill/SuperSlicer #3032: Arachne inconsistency.

Dx-Dx-Dx, 2022-08-05:
### What happened?

I find SS's Arachne unusable:
[image]
PrusaSlicer looks much better:
[image]

Settings between SS and PS seems to be the same.

### Project file & How to reproduce

[TriHorn Duct Default.zip](https://github.com/supermerill/SuperSlicer/files/9268428/TriHorn.Duct.Default.zip)

### Version

Version: 2.5.59.0 Build: SuperSlicer_2.5.59.0_2022-08-05

### Operating system

Win7 64

supermerill, 2022-08-05:
working the same if I use the code only avaialble at the alpha3 time.
There is something in prusa fix or (more probable) in my fixes that mess it up.

supermerill, 2022-08-06:
if you deactivate the overhangs detection, it returns back to normal.

supermerill, 2022-08-06:
should be fixed in the next nightly (currently ubuilding)
forgot to tag it.

closed 2022-08-08 by Dx-Dx-Dx

Dx-Dx-Dx, 2022-08-08:
Tnx, looks good. How it is better than classic is not apparent but arachne at least can be faster.
[image]

locki-cz, 2022-09-11:
It is fixed?

[image]

supermerill, 2022-09-12:
> It is fixed?

what?