supermerill/SuperSlicer #3033: Manual control of changing print order when printing multiple instances simultaneously

ppp475, 2022-08-05:
**Is your feature request related to a problem? Please describe.**
When I print multiples of the same objects, and I use the Arrange button to lay them out on the bed, I've found it appears to be random where the first instance is placed in relation to the other instances. In the attached images, Instance 1 is the second from the left. This causes the printer to print the second item in the row first, then backtrack to the first, then move over the already printed layers to get to the third and fourth. This also can happen with items arrayed in the Y axis, where the printer can print the middle item first and the opposing corner items last, potentially dragging the nozzle over the existing layers or causing excessive stringing between parts. 

**Describe the solution you'd like**
Some way to either manually choose which items print first, like a print queue (Popup menu that highlights the item when you select it?), or an option that automatically set the order to options like "Right to left", "Top to Bottom or Back to Front", ect. 

**Describe how it would work**
Popup menu: Popup window similar to existing "Object Settings" popups, with a list of instances. When an object is selected on the list, it is highlighted on the bed, like the current selection highlight. The print order is from Top to bottom of the list, and you can reorder the list with arrow buttons that move the selected line up and down, or clicking and dragging. Possibly also add a "Queue Number" field that could be typed into and reorder parts accordingly like a wishlist.

Built in Option: Print Settings > Output Options > Print Order (possibly filed under Sequential Printing but that might be confusing). Just a dropdown menu that has options that provide an ordered way to print the parts. For example, Left to Right, Right to left, Top to Bottom, Bottom to Top, ect. Maybe combine them so it's "Left to right, Top to bottom".

**Describe alternatives you've considered**
Manually reordering the parts by selecting the instances and moving them to be in the correct order. Has worked with mixed results, anything more than 3 or 4 items it seems to change which one it starts from.

**Additional context**
Current:
[image]

Proposed:
[image]

neophyl, 2022-08-05:
It prints them in the order that they are listed in on the tree in the right panel.  You can use the shortcut "e" to highlight which object is which if you want to place them in a specific place.

Admittedly the auto arrange could use some work to actually arrange them in print order but at least you can manually do it if thats something that really bothers you.

ppp475, 2022-08-05:
Yeah, that was basically what my workaround was. Manually fixing it is fine, but it'd be nice to not have to do it in the first place. Definitely more of a "nice to have" feature as opposed to a critical one.

SuperDodge, 2023-05-21:
> It prints them in the order that they are listed in on the tree in the right panel. You can use the shortcut "e" to highlight which object is which if you want to place them in a specific place. [...]

This is not the behavior I am experiencing.  I can move these things around all I want and when I slice they change and print in whatever order they want.