supermerill/SuperSlicer #3035: One one perimeter on top surfaces grayed out with Arachne selected

be221b, 2022-08-06:
### Version
2.5.59.0 (Nightly)

### Operating system type + version
Win11 x64

### Behavior
Not sure if this is a bug or a feature, but "One one perimeter on top surfaces" grayed out with Arachne selected. Can still select only one perimeter on first layer though.

supermerill, 2022-08-06:
not compatible.
Arachne is a different way of doing periemters.
I may integrate it into arachne later on, if it's possible.

be221b, 2022-08-06:
Interesting. I can use a height modifier to manually do it on a top layer so hopefully it's not too hard to implement eventually. Thanks for all the good work!

dderg, 2023-04-01:
the option is grayed out, but if you turn arachne off, then enable only 1 perimeter on top layer, and then turn arachne back on, it does generate g-code with 1 perimeter top layer. I had the opposite problem not knowing how to turn it off.

justr0st, 2023-04-03:
> not compatible. Arachne is a different way of doing periemters. I may integrate it into arachne later on, if it's possible.

It works however on the topmost layer if it was activated for classic perimeter generator and then switched to Aracne:
[image]

supermerill, 2023-11-27:
nice, i'll tag that to merge it in the 2.7