supermerill/SuperSlicer #3036: Infill density modifier/variable infill density

ElliottBridge, 2022-08-06:
Have some parts of your model that need strength and others that just need form (aesthetic)? Want to change centre of mass or inertia?
Want to do either of these without changing external geometry? Well, currently there doesn't really seem to be a way.

Well my idea would be to have variable infill density!

In my mind this could work one of two ways (even both since they could both be useful depending on what you are trying to achieve.

1. Simple equations (coordinate geometry) that can describe how %infill changes along the x, y, and z axis.

2. Using modifier shapes like with supports, etc. Probably much easier to implement since the %infill changes would be in discrete zones, rather than being graduated as described by an equation which probably requires a rework on how infill is generated altogether.

Maybe this is easier to implement with some infills rather than others such as homogeneous ones like gyroid? If so that's totally fine. Although this could be quite innovative in slicing technology, I do understand it may be hard to implement, but I do feel as though it would be worth it, particularly in practical design, for instance robot arm parts where the moment of inertia needs to be as close to the centre of rotation as possible, etc, etc.

supermerill, 2022-08-06:
you can have different infill densities with modifiers.

Do you want a feature that can place a centre of gravity somewhere?
The easiest way may be to remove some rectilinear infill lines to move it as much as possible next to the wanted location.
It has to be done after al perimeter & infill creation to be able to compute it.

ElliottBridge, 2022-08-07:
> you can have different infill densities with modifiers.
>
> Do you want a feature that can place a centre of gravity somewhere?
> The easiest way may be to remove some rectilinear infill lines to move it [...]

Can it really? Can't believe I missed that 😂. The idea to choose centre of gravity and for it to self adjust would be amazing though, I love that!