supermerill/SuperSlicer #3038: App not opening on Mac

jimmyD30, 2022-08-07:
### What happened?

Hello, the latest app version is not opening on Mac.

### Project file & How to reproduce

Install app onto Mac then try to open it.

### Version

2.4.60.2

### Operating system

Mac OS 15

### Printer model

Custom with Klipper

hairydalek, 2022-08-07:
I get the following error:
Internal error: basic_string

This is when the splash screen is displayed
[image]
.

supermerill, 2022-08-07:
intel version on intel mac?

jimmyD30, 2022-08-07:
> intel version on intel mac?

For me, yes.

hairydalek, 2022-08-07:
Yes, Intel Mac and Intel version.

jimmyD30, 2022-08-07:
So latest release version 2.3.57.12 works, but not the latest daily. Here is my macbook pro info:
[image]

jimmyD30, 2022-08-07:
This is the error i get with 2.4.58.3:
[image]
:

supermerill, 2022-08-23:
should be fixed in latest release, can you test?

jimmyD30, 2022-08-24:
> should be fixed in latest release, can you test?

Hi, yes, will try in a day or two, thank you!

jimmyD30, 2022-08-24:
> should be fixed in latest release, can you test?

She opens and slices (did not try gcode output yet), but I get this error message:
[image]

boksbox, 2023-01-08:
SuperSlicer_2.4.58.5_macos_220924 opens but refuses to stay open.  Mac OS 10.6.2

closed 2024-02-11 by supermerill