supermerill/SuperSlicer #3040: Spiral Vase Defect

CCS86, 2022-08-07:
### What happened?

I'm not sure what setting might be causing this aligned "dive" inwards. 

[image]

My first thought was these two settings (which I think should be treated as zero when spiral vase is enabled, so you don't need a "vase" machine config):

[image]

But, after zeroing them, the defect is still there.

### Project file & How to reproduce

[Plunger.3mf.txt](https://github.com/supermerill/SuperSlicer/files/9277192/Plunger.3mf.txt)

### Version

2.4.58.3

### Operating system

Windows 10 x64

### Printer model

UMO+

supermerill, 2022-08-07:
That's an artifact from how the spiral vase is generated.
it's still a hacky post-process.
That means that is still slices the object on a layer -per layer bassis, and then distord them on the z axis.

on the top, you have a chamfer.
each of these layers is a perfect circle.
but each layer is a smaller circle.
when the algo has to join the circle together, it has to "lunge" from one to another, hance what's you are seeing.

CCS86, 2022-08-08:
> That's an artifact from how the spiral vase is generated. it's still a hacky post-process. That means that is still slices the object on a layer -per layer bassis, and then distord them on the z [...]

Is it possible to create a setting which slices at a much thinner layer height than you intend to print at? This way you could make more frequent and smaller corrections. I would gladly wait longer for slicing to avoid a big once-per-layer defect like this.

supermerill, 2022-08-09:
You can already reduce your layer height.
If you want to do a half-circle each layer, it's a bit complicated, as you have to find a spot where you can jump safely.
I shouldn't be extra hard to implement, in a first thought. Still not straightforward.

CCS86, 2022-08-11:
> You can already reduce your layer height. If you want to do a half-circle each layer, it's a bit complicated, as you have to find a spot where you can jump safely. I shouldn't be extra hard to implement, in a first thought. Still not straightforward.

Yes, you can reduce layer height, but there is a practical limit to that. 

For the sake of print time and nozzle size, you will come up with a desired layer height.

The enhancement could be a "Spiral Refinement Multiplier". Setting this to 4x would slice the model at 1/4 the printing layer height, then make the XY correction 4 times per loop, instead of one. 

If you wanted to print at 0.2mm layer height, then set a 10x Refinement Multiplier, it would slice the model at 0.02mm, and make 10 XY course corrections per actual layer/loop.

SytanSD, 2023-01-14:
Hoping to see this fixed myself. I know NOTHING of coding, but I do know how many problems this has been causing me with surface finishes on pots and vases and such. Here you can see the seam introduced on a model I designed and printed for somebody to use functionally. The seam is far more visible in person, but I do not have a photo of it ATM

[image]

and down below here you can see an extremely dramatic recreation using a sphere. I am sure I am super wrong in saying this, but I feel like this would be a simpler problem to solve than a lot of previous issues. I am sure I have no idea what I am talking about in this matter, and I apologize for my lack of knowledge. I just know my personal experience with the problems this has caused me.

[image]