supermerill/SuperSlicer #3042: Introduce filament profile change on colour change

paulstimpson, 2022-08-07:
**Is your feature request related to a problem? Please describe.**
Print defects can occur after the {+} button next to the layer height is clicked to add a colour change due to the slicing settings for the original filament still being used

**Describe the solution you'd like**
As well as changing the colour of the filament in the preview, I would like to be able to select a new filament profile to pick up the new nozzle temperature, flow and filament overrides for the new filament, such as retraction to be used for the subsequent layers.

**Describe how it would work**
As well as letting the user choose the colour they want shown in the preview for layers above the change, there would be an option to select a new filament profile so those layers are sliced with appropriate settings for that filament. An automatic re-slice of those layers would then be commanded to generate the correct GCode 

**Describe alternatives you've considered**
I have been unable to come up with an alternative

**Additional context**

Pavulon87, 2022-08-10:
Set appropriate number of extruders/filaments and change the material according to the extruder.
[image]

supermerill, 2022-08-10:
you can switch extruder instead of the pause.
Don't forget to setup your M600 in the toolchange gcode.