supermerill/SuperSlicer #3043: arachne is make under extrusion width

wonsuhna1, 2022-08-07:
### What happened?

[image]

[image]

0.6 nozzle, 0.65~0.7 extrusion width
like that image,
perimeter is under extrusion width problum.
Doesn't match the shell

what happend?

It has been confirmed that this problem does not occur with prusaslicer.

### Project file & How to reproduce

[temp Mo.zip](https://github.com/supermerill/SuperSlicer/files/9277654/temp.Mo.zip)

### Version

nightley

### Operating system

windows10

supermerill, 2022-08-07:
there is currently a bug in the arachne: the widths are wrong. fixed in next nightly, can you check it tomorrow?

wonsuhna1, 2022-08-07:
> 현재 아라크네에 버그가 있습니다. 너비가 잘못되었습니다. 다음 밤에 수정, 내일 확인할 수 있습니까?

yes, - thankyou!

pemerick07, 2022-09-07:
When I tried Arachne it required an EM of 1.2 to get to the same measured part weight that I got from Classic with an EM of 1.075. I found this issue and tested the 05585fd nightly which seems to have resolved it - Thanks!

Note the Used Filament.

**2.5.59 Classic with 1.075 EM**
[image]

**2.5.59 Arachne with 1.2 EM**
[image]

**Nightly with 1.2 EM**
[image]

**Nightly with 1.075 EM**
[image]

closed 2024-02-11 by supermerill