supermerill/SuperSlicer #3046: Arachne , infilled bottom layer with monotnic/filled

jesterhead82, 2022-08-08:
### What happened?

On todays nightly, active one perimeter in first layer, bottom monotonic filled.

Can be fixed by chosssing monotonic fill (so small area gets filled), or by deselecting single perimeter on first layer.

[image]
[Uploading 3D4U_Coffee_Clip_Standard_V2.zip…]()

### Project file & How to reproduce

Small features dont get filled by "filled" infill

### Version

2.5.59.0 nigtly

### Operating system

Win10

### Printer model

Other

supermerill, 2022-08-08:
I think your link is broken. can you try again to add it?

jesterhead82, 2022-08-08:
[3D4U_Coffee_Clip_Standard_V2.zip](https://github.com/supermerill/SuperSlicer/files/9280415/3D4U_Coffee_Clip_Standard_V2.zip)

sure

Dx-Dx-Dx, 2022-08-08:
It seems all filled infills are no longer.
[image]
Especially noticeable on concentric infill - in classic mode both concentric and concentric filled are almost the same. In arachne mode concentric infill is arachned and looks good but concentric filled is not and not filled.

Dx-Dx-Dx, 2022-08-08:
[heartgear8.zip](https://github.com/supermerill/SuperSlicer/files/9283605/heartgear8.zip)
Version: 2.5.59.0
Build: SuperSlicer_2.5.59.0_2022-08-07

supermerill, 2022-08-09:
@Dx-Dx-Dx fixed

closed 2024-02-11 by supermerill