supermerill/SuperSlicer #3047: Aligned Seem doesn't work correctly?

TjardF, 2022-08-08:
### Version
Version: 2.4.58.3
Build: SuperSlicer_2.4.58.3_2022-07-13

### Operating system type + version
Mac OS Catalina 10.15.7 (19H1030)
### Behavior

I inserted this Part to SuSi, its from Voron Design, Trident Build current Version and sliced it, my default setting is aligned Z Seam, but the seam doesn't align. not at all. in the full print.
In each hole etc

[image]

#### Project File (.3MF) where problem occurs
[zSeam.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9283237/zSeam.3mf.zip)

locki-cz, 2022-08-09:
In latest PS 2.5.0 Alfa 3 are seams so nice! Also drawed seam is aligned very nice. Can it be imported to SS?

supermerill, 2022-08-09:
> In latest PS 2.5.0 Alfa 3 are seams so nice! Also drawed seam is aligned very nice. Can it be imported to SS?

Will be merged

locki-cz, 2022-08-10:
still buggy - not fixed in today version
[image]

[image]

supermerill, 2022-08-10:
It work on the provided project file.
you can use a seam-sphere if you want a precise control over it.
also, did you try the "contiguous" ?

locki-cz, 2022-08-10:
Yes used Contiguous. Arachne destroy it, also my drawed line is ignored :) 
[[a]_stealthburner_main_body.zip](https://github.com/supermerill/SuperSlicer/files/9299937/a._stealthburner_main_body.zip)
Seam-sphere

supermerill, 2022-08-10:
> Seam-sphere? Googled but not found anything related.

https://github.com/supermerill/SuperSlicer/search?q=seam+sphere

right-clic -> seam

locki-cz, 2022-08-11:
Oh this is interesting but useless in this case, it make seam on both sides :(

[image]

in prerelase it is stil broken, ignoring drawed position
[image]
[image]

But I must say, other is alligned really awesome way!

supermerill, 2022-08-11:
I forgot. Tu use the aligned, you should only put a little painted seam in the first layer(s). It will follow on from here.

locki-cz, 2022-08-11:
I dont know about this! There sould be something like tips about how to use some feacures :)
Perimeter generator: classic
[image]

Perimeter generator: arachne  is broken :( 
[image]

closed 2024-02-11 by supermerill