supermerill/SuperSlicer #3051: SuperSlicer Model not loading on zero Z plane

BJONES673, 2022-08-09:
Super Slicer Version 2.4 

### Operating system type + version
Windows 10 Professional, FLsun Super Racer Printer, Stock except for BMG extruder, Dual Metal Heatbreak. 

### When I load a 3D model, it shows graphically in Superslicer that the model is flat on the base of the printer. However, when you look at the Z location, it is actually above the zero plane. For instance, I loaded a test model and the model shows it is flat on the printer bed (even shows the first layer on the bed when slicing). However the Z level shows it at a positive 27.3mm. If I change the Z to a zero level, the model then shows half way below the bed level. However, if I print the model, it does print correctly with th offset at the elevated position. Where it becomes an issue is on tall models, especially since this is a delta printer. This is on all models. When I load the exact same stl's in Cura or Ideamaker, they are correctly seated on the print bed at a zero Z location. I have looked at every setting in superslicer and cannot figure it out. I loaded the models in meshmixer and see no errors. I am at a loss of why this happening and how to fix it. Thank everyone for any help. I am newer to superslicer, so I may be missing something
=

_Is this a new feature request?_ No

supermerill, 2022-08-09:
your project file?

BJONES673, 2022-08-09:
Sorry - thought I dropped those in. Here are two examples - had to load the zip files, STL's as it seems STLs cannot be loaded 
[cartesian-printer-test-piece20190228-11944-19qd0lx.zip](https://github.com/supermerill/SuperSlicer/files/9294
[scarecrow.zip](https://github.com/supermerill/SuperSlicer/files/9294155/scarecrow.zip)
153/cartesian-printer-test-piece20190228-11944-19qd0lx.zip)

BJONES673, 2022-08-09:
[cartesian-printer-test-piece20190228-11944-19qd0lx.zip](https://github.com/supermerill/SuperSlicer/files/9294166/cartesian-printer-test-piece20190228-11944-19qd0lx.zip)

supermerill, 2022-08-09:
> However, when you look at the Z location, it is actually above the zero plane.

The slicer will use the origin of your object. It has to move it up to lay it on the bed.

> However, if I print the model, it does print correctly with th offset at the elevated position.

yes, working as expected.

BJONES673, 2022-08-09:
Thank you. So the reason this is an issue is that when I print a tall model it artificially hit the limit of the delta printer height and the problem compounds on tall prints (as it starts even higher). I am not sure why this happens, but there is something weird about the location of the print in terms of the a height in relation to the printer. 

On the scarecrow print, the top of the print was complete spaghetti - I tried it 3 times and all failed at the exact same spot - so it's not a physical printer issue. I guess I could artificially raise the limit of the printer height in superslicer, but that sounds a bit risky. Also, why does cura and ideamaker both make this correction and keep the z height at zero? 

If this is something the super slicer has an issue with, is there way I can correct the file to change the point of origin to true zero at the Z height? It appears to be correct when I load it into Meshmixer, but when I export the file, it still show the same elevated Z point in Superslicer. Sorry for the rudimentary questions, learning as I go... 

Thank you again for the help

supermerill, 2022-08-09:
> Also, why does cura and ideamaker both make this correction and keep the z height at zero?

They hide it. It's done internally

> If this is something the super slicer has an issue with

there is no issue. Your stl load and slice correctly with my config & printer.

>  is there way I can correct the file to change the point of origin to true zero at the Z height?

yes with blender, freecad, or any 3D editor. it's relative anyway, it has no consequence.

BJONES673, 2022-08-09:
Hmmmm - then the issue with the top of print failing at the exact same point is something else. It's not the set-up, as even the scaled up print is within the limits of the print (as validated by superslicer). I forgot to mention that this print was scaled to 400%. At the original size, there is no print issue. But at 400%, the top of the print fails

BJONES673, 2022-08-09:
I am going to print it with ideamaker at the same 400%, my gut tells me it will print correctly, but I need to test it. If it works, then there is an issue at the higher levels with the superslicer, if not then it has to be something with the printer