supermerill/SuperSlicer #3055: tool tip for hole compensation is misleading/incorrect

telepresencebot2, 2022-08-10:
### What happened?

Tooltip reads as follows:

> The conves holes will be grown / shrunk in the XY plane by the configured value (negative = inwards, positive = outwards, should be negative as the holes are always a bit smaller irl). This might be useful for fine-tuning hole sizes.

"should be negative as the holes are always a bit smaller irl" is incorrect. Positive values increase hole size to counter irl shrinking. A negative number makes an undersized hole in a print even more undersized.

It should read "should be positive" instead.

### Project file & How to reproduce

under print settings > slicing > modifying slices, hover over Vertical Hole shrinking compensation > XY compensation for the tool tip

### Version

2.5.59.0

### Operating system

windows 10

### Printer model

voron 2.4

supermerill, 2022-08-10:
inward = inside the part. it means that the hole will be bigger.

telepresencebot2, 2022-08-10:
> inward = inside the part. it means that the hole will be bigger.

Incorrect, I have tested. A negative value shrinks interior holes, a positive value makes interior holes larger

telepresencebot2, 2022-08-10:
I just pulled a part off my printer with a value of positive .05mm.
an interior hole with diameter 8.8mm has a measured diameter of 8.78 with that value.
a part printed from the same model and profile, but a value of negative .04mm has a measured size of 7.9mm.

all other settings were the same, this compensation value was all I adjusted.

telepresencebot2, 2022-08-10:
inwards = towards the center of the hole

telepresencebot2, 2022-08-11:
[image]
Evidence that I have actually tested this.
The far left print was with -.2mm compensation and you can visibly see that the smallest hole is significantly smaller on that print than any other. (1.9mm diameter for a 3.2mm modeled hole)

From left to right the compensation values are
-.2mm, -.04mm, 0, .05mm

telepresencebot2, 2022-08-11:
it can also be verified in the gcode previe
![negative_compensation](https://user-images.githubusercontent.com/8864900/184046064-7de282b9-f54a-4c42-80a2-08e
[image]
b898722f6.png)
w

supermerill, 2022-08-11:
You're right.
The behavior was correct in the 2.3.56.7 and 2.3.57.7.
But it isn't in the 2.3.57.12
I wonder why... will investigate.

macchinette, 2022-08-13:
> You're right. The behavior was correct in the 2.3.56.7 and 2.3.57.7. But it isn't in the 2.3.57.12 I wonder why... will investigate.

It has changed I don't know when, but it seems to be more consistent since if I want to remove from the elephant foot I use negative values and if I want to enlarge the holes I use positive values.
I like it like this, the only thing that there is always the problem that detects holes in all the parts that have a closed perimeter and empty in the middle, I had already told you about it (https://github.com/supermerill/SuperSlicer/issues/1959)

neophyl, 2022-12-14:
The point being is that you arent enlarging or shrinking the HOLE, you are shrinking the material around the hole which results in the changed size.  Thats why the original operation makes sense, what you propose and the ways its broken now then the 2 things which should do the same thing actually do opposite.

macchinette, 2022-12-14:
Hi, I understand little about it, but for me the function must do that if the printer makes me the smaller round holes I want to be able to enlarge them with a positive value and if it makes me the holes wider I want to be able to narrow them with a negative value.  The important thing is that this function is applied only and exclusively to the round holes and not to the geometric shapes of closed and empty walls inside. 
Thank you