supermerill/SuperSlicer #3056: Arachne wipe moves outside the external perimeter

Lunerwalker2, 2022-08-11:
### What happened?

When using the arachne engine, I found that seams on the external perimeters suddenly became massive. After watching another print while it completed the outer perimeter, I saw that the wipe upon completion traveled outside the perimeter, causing filament to be placed on the seam.

A look at the slicer showed that the nozzle goes out into the air when it wipes with arachne.
[image]

This is different from the classic engine, which does not go so far out. 
[image]

This is causing very obtrusive seams to form, making the holes unusable. 
[image]

### Project file & How to reproduce

[Outer Hub - HTD-5mm-25T-Pulley(9).3mf.zip](https://github.com/supermerill/SuperSlicer/files/9305262/Outer.Hub.-.HTD-5mm-25T-Pulley.9.3mf.zip)

### Version

2.5.59.0

### Operating system

Linux Mint 20.3 Cinnamon

### Printer model

Ender 3 Pro

tome9111991, 2022-08-11:
i think i have similar issue
[image]

[wipe.zip](https://github.com/supermerill/SuperSlicer/files/9308021/wipe.zip)

Lunerwalker2, 2022-08-14:
Weirdly enough, the most recent arachne prelease seems to not have this issue anymore.

TexasTomBF, 2022-09-04:
I can clearly see the same bug, making arachne engine not attractive to use yet

Chukozor, 2022-09-04:
For the time being I use the arachne engine with only the "extra wipe for external perimeters" activated. There is no issue with it, but indeed I'm looking for the fix of the "wipe inside at start/end". @TexasTomBF

closed 2024-02-11 by supermerill