supermerill/SuperSlicer #3058: Could the new "contigous" seam follow the seam sphere?

hagan66, 2022-08-11:
I just love how smooth the new contigous seam is, I just want to be able to place it where I want without rotating the model. 

Thanks, and fantastic work btw!

/robert

supermerill, 2022-08-28:
I don't understand the issue.
The seam sphere has priority over the seam algorithm.
Can you explain more? with pictures, or/and project file?

hagan66, 2022-08-28:
OK, what I mean is that the contiguous seam is much more in-line-smooth than the seam-sphere-seam. 
I would like to place the contiguous seam in any direction, like with a seam sphere. :-)

[image]

[image]

supermerill, 2022-08-28:
Understood.

Did you try to put a little dot of seam paint (the left widget) just at the very bottom of where you want it to go?

[image]

hagan66, 2022-08-28:
I tried it now. It works with ”aligned” but not with ”contiguous”.

closed 2024-02-11 by supermerill