supermerill/SuperSlicer #3059: Question - 1 perimeter on top disabled with Arachne

neophyl, 2022-08-11:
### What happened?

The one perimeter on bottom layers is selectable (and works) with both Classic and Arachne but the option for 1 perimeter on the Top layer becomes greyed out when Arachne is selected.  I also never slices with 1 perimeter.  Why the difference ?

### Project file & How to reproduce

Just open any new project and load any object.

### Version

2.5.59

### Operating system

Win 10

### Printer model

Mk3 but doesnt matter

nielsvz, 2022-08-11:
From the release notes, it's not implemented yet:
> Arachne perimeter generation from PrusaSlicer, from Cura, based on a paper by Tim Kuipers. It creates variable extrusion width perimeters, to avoid gapfill.
It won't have many of the customization available for perimeters, as everything needs to be re-implemented from scratch. Currently, the fuzzy skin and the overhangs are done. The one-loop perimeter may follow, but it's harder as arachne doesn't create only loops.

neophyl, 2022-08-11:
Yes I know but seeing as its available on the 1 loop on bottom perimeter option hence the question.  I would have expected both to be disabled or both to be available.

ocochodzi111, 2022-08-11:
It is really crucial option for me and i think many of people using it. Uniform look of new and already produced parts etc.

supermerill, 2022-08-11:
it's not one loop on bottom but one loop on first layer.
much much easier.

DaStivi, 2022-08-12:
Hello not sure if this is realted to "one loop on top layer" and its not supported in arachne, i'll tried to switch back to classis and also disabled (the greyed out) option anyway... but looks like somehow the TOP "infill" isn't applied here?!

classic left, arachne right
[image]