supermerill/SuperSlicer #3060: 2.5.59.0 Crash during launch (MACOS)

d-nx, 2022-08-11:
### What happened?

When attempting to launch SuperSlicer.app I get a crash report containing the following:

`Process:               SuperSlicer [26466]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.5.59.0_2022-08-11 (SuperSlicer_2.5.59.0_2022-08-11)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [26466]
User ID:               502

Date/Time:             2022-08-11 14:40:11.020 -0400
OS Version:            Mac OS X 10.14.6 (18G9323)
Report Version:        12
Anonymous UUID:        39B03C67-F739-452A-C956-F8F8885EDBE1

Time Awake Since Boot: 1400000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @executable_path/libzstd.1.dylib
  Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
  Reason: no suitable image found.  Did find:
	/Applications/SuperSlicer.app/Contents/MacOS/libzstd.1.dylib: cannot load 'libzstd.1.dylib' (load command 0x80000034 is unknown)
	/Applications/SuperSlicer.app/Contents/MacOS/libzstd.1.dylib: stat() failed with errno=1
	/Applications/SuperSlicer.app/Contents/MacOS/libzstd.1.dylib: cannot load 'libzstd.1.dylib' (load command 0x80000034 is unknown)
	/Applications/SuperSlicer.app/Contents/MacOS/libzstd.1.dylib: stat() failed with errno=1

Binary Images:
       0x10c305000 -        0x10e4c8ff7 +com.merill.slic3r/ (SuperSlicer SuperSlicer_2.5.59.0_2022-08-11 - SuperSlicer_2.5.59.0_2022-08-11) <96E9B703-ACE5-3446-B2D3-C7FB86E377C7> /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
       0x1181bc000 -        0x11822670f  dyld (655.1.1) <4842B61A-B868-3146-9CA9-5582C4198B25> /usr/lib/dyld
    0x7fff2e99c000 -     0x7fff2ede2fff  com.apple.CoreFoundation (6.9 - 1575.405) <04F56DFB-C410-31B8-8AE9-7D2024DB6E22> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
    0x7fff312dd000 -     0x7fff3136cfff  com.apple.framework.IOKit (2.0.2 - 1483.260.5) <C00462C2-9CF3-3E3E-BD1E-B57AF6AB562E> /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
    0x7fff3864f000 -     0x7fff3865effb  com.apple.opengl (17.7.3 - 17.7.3) <6CD93FE7-C029-316B-A08D-54E5EC9F0830> /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
    0x7fff57aa8000 -     0x7fff57aa9ffb  libSystem.B.dylib (1252.250.1) <B2E16B83-D383-330A-8BD3-39526248EDFD> /usr/lib/libSystem.B.dylib
    0x7fff57cdf000 -     0x7fff57cecfff  libbz2.1.0.dylib (38.200.3) <272953A1-8D36-329B-BDDB-E887B347710F> /usr/lib/libbz2.1.0.dylib
    0x7fff57ced000 -     0x7fff57d40ff7  libc++.1.dylib (400.9.4) <9A60A190-6C34-339F-BB3D-AACE942009A4> /usr/lib/libc++.1.dylib
    0x7fff586ad000 -     0x7fff5879dfff  libiconv.2.dylib (51.200.6) <2047C9B7-3F74-3A95-810D-2ED8F0475A99> /usr/lib/libiconv.2.dylib
    0x7fff58a4e000 -     0x7fff58a66ff3  liblzma.5.dylib (10.200.3) <E1F4FD60-1CE4-37B9-AD95-29D348AF1AC0> /usr/lib/liblzma.5.dylib
    0x7fff5a152000 -     0x7fff5a164ff7  libz.1.dylib (70.200.4) <B048FC1F-058F-3A08-A1FE-81D5308CB3E6> /usr/lib/libz.1.dylib

Model: iMac14,2, BootROM 431.140.6.0.0, 4 processors, Intel Core i5, 3.2 GHz, 24 GB, SMC 2.15f7
Graphics: kHW_NVidiaGeForceGT755MItem, NVIDIA GeForce GT 755M, spdisplays_pcie_device, 1 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x0000, -
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x0000, -
Memory Module: BANK 0/DIMM1, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
Memory Module: BANK 1/DIMM1, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x111), Broadcom BCM43xx 1.0 (7.77.61.3 AirPortDriverBrcmNIC-1305.12)
Bluetooth: Version 6.0.14d13, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Mercury Electra 6G SSD, 500.11 GB
USB Device: USB 3.0 Bus
USB Device: TR-004
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: Audient iD14
USB Device: nanoKONTROL2
USB Device: USB2.0 Hub
USB Device: iLok
USB Device: USB2.0 Hub
USB Device: iLok
USB Device: Keychron K4
USB Device: Keystation Mini 32 MK3
Thunderbolt Bus: iMac, Apple Inc., 23.10
`

### Project file & How to reproduce

Launch SuperSlicer v2.5.59.0 on MAC

### Version

2.5.59.0

### Operating system

MacOS

supermerill, 2022-08-11:
seems like the included zstd can't be found, or is the wrong one.
Also the same with the 2.4.58.4
I have published again the intel-only version, but these aren't signed, so you'll have some warning.

Creo2005, 2022-08-12:
In my all is OK with latest night build for macos

supermerill, 2022-08-12:
@Creo2005 all is ok with what version on what hardware?

Creo2005, 2022-08-12:
> @Creo2005 all is ok with what version on what hardware?

Hackintosh :) 
[image]

WouterE99, 2022-08-15:
Hello, 

I just downloaded the latest version and my Mac is running the latest version (Monterey 12.5) but I still get the same error.

Crash report:

Process:               SuperSlicer [2037]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.4_2022-08-10 (SuperSlicer_2.4.58.4_2022-08-10)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-08-15 18:28:25.4011 +0200
OS Version:            macOS 12.5 (21G72)
Report Version:        12
Bridge OS Version:     6.6 (19P6064)
Anonymous UUID:        04CDDCE8-3618-6E73-01F8-7281D5546290

Sleep/Wake UUID:       8D8026F3-6320-457C-BC58-821EC85DD277

Time Awake Since Boot: 1100 seconds
Time Since Wake:       301 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib'
Referenced from: '/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer'
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x11b35b98e __abort_with_payload + 10
1   dyld                          	       0x11b371a72 abort_with_payload_wrapper_internal + 80
2   dyld                          	       0x11b371aa4 abort_with_payload + 9
3   dyld                          	       0x11b31f699 dyld4::halt(char const*) + 615
4   dyld                          	       0x11b31b5b5 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3995
5   dyld                          	       0x11b31a4e4 start + 388

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b61ef638  rdx: 0x00007ff7b61efaa0
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7b61ef680  rsp: 0x00007ff7b61ef638
   r8: 0x00007ff7b61ef6a0   r9: 0x0000000000000000  r10: 0x0000000000000075  r11: 0x0000000000000246
  r12: 0x0000000000000075  r13: 0x00007ff7b61efaa0  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x000000011b35b98e  rfl: 0x0000000000000246  cr2: 0x000000011b321da4
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x11b315000 -        0x11b380fff dyld (*) <f71fb3ca-5fcc-3577-9457-b047888a46d1> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
[... 113 lines cut ...]
Graphics: Intel Iris Plus Graphics 645, Intel Iris Plus Graphics 645, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Display: 27B2G5, 1920 x 1080 (1080p FHD - Full High Definition), MirrorOff, Online
Display: DELL P2719HC, 1920 x 1080 (1080p FHD - Full High Definition), MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 4 GB, LPDDR3, 2133 MHz, SK Hynix, H9CCNNNBJTALAR-NVD
Memory Module: BANK 2/ChannelB-DIMM0, 4 GB, LPDDR3, 2133 MHz, SK Hynix, H9CCNNNBJTALAR-NVD
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x870), wl0: Jul 16 2021 18:25:13 version 16.20.328.0.3.6.105 FWID 01-30be2b3a
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: hub_device
USB Device: 4-Port USB 3.0 Hub
USB Device: USB31Bus
USB Device: hub_device
USB Device: MasterKeys Pro M Intelligent RGB
USB Device: composite_device
USB Device: 4-Port USB 2.0 Hub
USB Device: T2Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

CSWP-Spark, 2022-08-17:
Install brew: /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"

Install zstd: brew install zstd

WouterE99, 2022-08-17:
This worked! Thanks!

jk257, 2022-09-03:
Had this same issue and "Install zstd: brew install zstd" solved it for me as well. Thanks!