supermerill/SuperSlicer #3062: [MacOS Arm] Not able to use post-processing scripts anymore

bistory, 2022-08-12:
### What happened?

I have two processing scripts in my configuration : The first is an executable and the second is a python script.
They used to work in the previous release but now I get an error when I try to launch them.

The curious thing is : I call them with the tilda shortcut (~/path/to/scripts/script.py) and now SuperSlice is throwing me the following error : 
`The configured post-processing script does not exist: ~/Projects/SuperSlicer/FastSQV.py`

So I changed the path to use the absolute path and I get this error : 
```
Post-processing script /Users/thomaslete/Projects/SuperSlicer/FastSQV.py on file /var/folders/fq/r82k4hd94y35k4j8wn3fxxg80000gn/T/.77282.gcode.pp failed.
Error code: 127
Output:
zsh:1: command not found: python
```

I tried using the absolute path of the python executable but it still doesn't work.

I think there is something broken with the execution environment...

### Project file & How to reproduce

1. Save the FastSQV.py file to your computer (https://github.com/K-Tett/Klipper-FastSQV/tree/main/superslicer)
2. Set it as post-processing script
3. Slice and try to save gcode file

### Version

2.5.59

### Operating system

MacOS 12.5

### Printer model

Voron 2.4 with Afterburner

supermerill, 2022-08-12:
I modified a bit the pot-process system, so it can add "python" itself if it detect a .py file. I may have forgot to detect if the "python" isn't already written....
Also, if the system can't find your file, it will try to find it from different location.
It should only makes it easier to configure, not harder :-/
I'll add some log to let users know what's going on.
Can you share your project, so i can try to reproduce?

try with `Projects/SuperSlicer/FastSQV.py` only.

tomohulk, 2022-08-13:
Not sure if we need a new issue for this, but same thing with Intel macOS, looks like a totally missing lib:

```
Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib'
Referenced from: '/private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer'
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)
```

supermerill, 2022-09-06:
I changed the python to python3 for linux & macos.

if you set your script to `python3 ~/Projects/SuperSlicer/FastSQV.py` does it already works?
if not, if you set your script to `python3 Projects/SuperSlicer/FastSQV.py` does it already works?
if not, if you set your script to `python3 /Users/thomaslete/Projects/SuperSlicer/FastSQV.py` does it already works?

bistory, 2022-09-06:
> I changed the python to python3 for linux & macos.
>
> if you set your script to `python3 ~/Projects/SuperSlicer/FastSQV.py` does it already works? if not, if you set your script to `python3 [...]

- `python3 ~/Projects/SuperSlicer/FastSQV.py` - Works
- `python3 Projects/SuperSlicer/FastSQV.py` - Fails (no such file or directory)
- `python3 /Users/thomaslete/Projects/SuperSlicer/FastSQV.py` - Works

👍

closed 2024-02-11 by supermerill