supermerill/SuperSlicer #3063: "Access (status) violation" on slicing anything 2.5.59.0 - no crash with console

aka13-404, 2022-08-12:
### What happened?

I use my calicat.stl which has worked for many versions before, and try to slice it. 
I get  "Access (status) violation" and no crash_log.txt
[image]
This is a clean install with configs freshly made, most stuff is on default. 
I get no crash when I run the same stl with the console version, so it must be the gui crashing? Not sure, what I may be doing wrong.
[image]

Appreciate your time as always, supermerill.

### Project file & How to reproduce

[Calibration Models.zip](https://github.com/supermerill/SuperSlicer/files/9329665/Calibration.Models.zip)

Simply import the stl, and press slice, you should get a crash. 

### Version

2.5.59.0

### Operating system

Windows 10

Arakon, 2022-08-12:
Has to be a specific setting, slices fine here with my printer's profile..

[image]

aka13-404, 2022-08-12:
I figured that much myself, that it is propably something in the settings, but how comes, that it crashes only in the gui mode? 
Must be some gui pereference, that is causing the issue, that would be the only reasonable assumption in that case. 

Do you mind sharing your gui preferences tab, I'll just copy them and see if anything changes.

aka13-404, 2022-08-13:
I can not reproduce it anymore, don't ask me how that works.

I have done the following steps: 
1) I backed up my config
2) I ran the setup wizard with some random printer
3) I compared the diff between the files, and changed them one after the other.

Results:
1) It was not the gui settings under perferences

2) Something along the highlighted options were responsible. I changed this whole section at once, so no idea what exactly it was. I tried setting it back to exactly how it is on the screenshot, but the crashing did not start again
[image]

Conclusion:
It was some 1/0 magic, sadly could not reliably reproduce. Closing the issue.

closed 2022-08-13 by aka13-404