supermerill/SuperSlicer #3064: MKS Thumbnail Support?

mlee12382, 2022-08-13:
Would it be possible to add thumbnail support for MKS TFT screens? Currently is seems as though Cura is the only platform that supports it via this plugin: https://github.com/Jeredian/mks-wifi-plugin It would be awesome if it could somehow be added to SuperSlicer. Not sure how difficult it would be to implement.

adripo, 2022-10-30:
Hi , I tried to create a PR to generate MKS TFT thumbnails, but I don't have much experience with c++. Can anybody check and help me? I added more details in the PR description. Thank you!

mlee12382, 2022-11-11:
That's awesome, I wish I had more programming experience. My programming skills are fairly limited. I'm glad someone started this addition to SS, hopefully you can get it ironed out and added to the next build, I know there are a lot of people that would love to use SS and still have thumbnails on the various printers that use the MKS hardware.