supermerill/SuperSlicer #3066: Superslicer White screen

Zybidk, 2022-08-13:
### What happened?

I have dowload SS 2.4.58.2 for Windows works about 10x  and when i start one more then come white screen, i have download SS 2.5.59 and have the same problem about 10x start and then White screen. SS 2.3.57 run without problem. Where is problem? 

### Project file & How to reproduce

[image]

### Version

2.3.57 now

### Operating system

Windows 11

### Printer model

V-minion, ender 3v2

Zybidk, 2022-08-26:
2.4.58 when i want go to card "print settings" then ss crash

supermerill, 2022-08-28:
can't reproduce and I can't open your picture.

Can you send me a zip of your configuration folder (when it's buggy)?
You can find it from help->show configuration folder.

Zybidk, 2022-08-28:
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/9439730/SuperSlicer.zip)
i hope you can finde where is problem.

Zybidk, 2022-08-28:
@supermerill i hope now come picture
[image] after i press print settings come white screen and ss 2.4.58 just crash

[image] ss 2.5.59 are white screen "3d view, sliced preview, gcode preview". i have wait about 2 min and nothing happend

supermerill, 2022-08-28:
Can't reproduce.
If you change the preferences -> GUI -> tab layout options to something else, does it works again?

Zybidk, 2022-09-02:
I have changed and no effekt. The same problem