supermerill/SuperSlicer #3067: Bed Texture gets Black after Mouse hover

TheRunnerPi, 2022-08-13:
### What happened?

Hi, my Bed Texture goes black after a hover over it with the mouse. After that it stays black. 

[image]
[image]

It happend after installing [2.4.58.3] but same problem with [2.4.58.4] and now even with the old [2.3.57.12]

I dont know how to fix it, but mabye someone has an idea :)

Thanks,
Daniel

### Project file & How to reproduce

[MK2.zip](https://github.com/supermerill/SuperSlicer/files/9331119/MK2.zip)

### Version

[2.4.58.4]

### Operating system

Windows 10

### Printer model

Custom

Seref, 2022-08-26:
Same here, do you by any chance have an AMD GPU ?

It seems like this was fixed on the n ormal Prusa-Slicer: https://github.com/prusa3d/PrusaSlicer/issues/8508#issuecomment-1220810939  would be cool if it could ported over.

Ojref, 2022-10-13:
This was fixed about a month and a half ago from today 10/23/22 in Prusaslicer. Considering the recent lag, it will probably be a while before its fixed here.

zhengyangliu, 2022-11-16:
Get same problem on RX6600XT.