supermerill/SuperSlicer #3071: Cannot Import config  PIF_45_Degree.ini or PIF_Standard.ini - Error "Failed loading the Config Bundle "": '=' character not found in line"

NikosSprocket, 2022-08-14:
### What happened?

Downloaded (save as) the PIF_45_Degree.ini and PIF_Standard.ini from
[https://github.com/AndrewEllis93/Ellis-PIF-Profile/blob/master/SuperSlicer/PIF_45_Degree.ini](url)
and tried to "Import Config" but I get the error:
"Failed loading the Config Bundle "": '=' character not found in line at line 8"
[image]

Windows Version:
Edition	Windows 11 Home
Version	21H2
Installed on	‎12/‎17/‎2021
OS build	22000.856
Experience	Windows Feature Experience Pack 1000.22000.856.0

### Project file & How to reproduce

[PIF_ini.zip](https://github.com/supermerill/SuperSlicer/files/9332744/PIF_ini.zip)

### Version

2.4.58.2, 2.4.58.4

### Operating system

Windows 11 - 21H2

### Printer model

Voron 2.4r2 - Bondtech LGX - Mosquito

neophyl, 2022-08-14:
neither of those ini files are actually ini files which is why its giving an error.  Open them up with any text editor and take a look for yourself.

They appear to be html from github.  For example the start of the PIF_standard file
<!DOCTYPE html>
<html lang="en" data-color-mode="auto" data-light-theme="light" data-dark-theme="dark" data-a11y-animated-images="system">
  <head>
    <meta charset="utf-8">
  <link rel="dns-prefetch" href="https://github.githubassets.com">
  <link rel="dns-prefetch" href="https://avatars.githubusercontent.com">
  <link rel="dns-prefetch" href="https://github-cloud.s3.amazonaws.com">
  <link rel="dns-prefetch" href="https://user-images.githubusercontent.com/">
  <link rel="preconnect" href="https://github.githubassets.com" crossorigin>
  <link rel="preconnect" href="https://avatars.githubusercontent.com">

Pavulon87, 2022-08-14:
You should use the RAW button:
https://raw.githubusercontent.com/AndrewEllis93/Ellis-PIF-Profile/master/SuperSlicer/PIF_45_Degree.ini

NikosSprocket, 2022-08-14:
Oooh no!
Sorry guys. I did see the <html> tags but I did not put it together at that moment that ini it's NOT html; Da!
I will close the case, with reason "user #@$#@%@@#$" lol

Thank you!
Nikos

closed 2022-08-14 by NikosSprocket