supermerill/SuperSlicer #3072: Strange travel after upgrade to 2.4.58.4

Shahriar57, 2022-08-14:
### What happened?

after upgrade to ver.2.4.58.4 travel movement is strange. as you can to see in picture 

 

### Project file & How to reproduce

[image]
[Pipe_Holder-Down.zip](https://github.com/supermerill/SuperSlicer/files/9332808/Pipe_Holder-Down.zip)

### Version

2.4.58.4

### Operating system

Windows 10

neophyl, 2022-08-14:
Your project has the following in the between layers section of your printer settings custom gcode
M104 S0
M140 S0
;Retract the filament
G92 E1
G1 E-1 F300
G28 X0
G1 Y300 F1500
M84

so you are momentarily setting the bed and extruder temp to zero on every layer (why ?) and also G28 X0 which is homing, which is what is causing some of your travel movement, the G1 Y300 F1500 command is causing more of it, and why are you disabling the steppers with M84 ?

To me it looks like everything that is in your between layer section should be in the End Gcode and that your end gcode looks like a copy of your start gcode.  Basically your printer profile is messed up.

Shahriar57, 2022-08-14:
it is end g-code, i replace it in wrong place, now everything working fine, thanks

closed 2022-08-28 by supermerill