supermerill/SuperSlicer #3073: MMU Tool Name extruder=extruder2 error

Kidabah, 2022-08-14:
### Version
2.5.59 Arachne

### Operating system type + version
Windows 11
### Behavior
* _Describe the problem_
*Since updating to Arachne version  SET_PRESSURE_ADVANCE ADVANCE=0 EXTRUDER=extruder2 gets sent when tool name = [blank] causing error in klipper
* _Steps needed to reproduce the problem_
* Have tool name = [blank] this will cause the error
* _Expected Results_
* SET_PRESSURE_ADVANCE ADVANCE=0 
* _Actual Results_
* SET_PRESSURE_ADVANCE ADVANCE=0 EXTRUDER=extruder2
  * _Screenshots if useful_
 Old tool change gcode
[image]
New tool change gcode
[image]

_Is this a new feature request?_
No
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

rsheng118, 2022-08-20:
i'm having the same issue.

Kidabah, 2022-08-20:
Unfortunately not much we can do. Until the error is fixed. I've gone back
to the mosaic version. But thanks for letting me know , Hopefully now there
are the 2 of us the issue may get addressed.

thatinkerer, 2022-08-22:
Same problem here with my Voron 2.4 and ercf. At the second Filament change inprint set_pressure_advance advance=extruder1 gives the toolname error and stops the print.

zuch95701, 2022-08-25:
Posted this same question on the commit where this change was implemented. Im having the same issue. Used a gcode replacement as an interim fix.  EXTRUDER=EXTRUDER1, EXTRUDER=EXTRUDER2 etc, replace with a simple space. Works for me but hope it gets fixed.

kovmarci86, 2022-10-08:
I just ran into this problem. I have my extruders named in the config, but I did not expect this to be included in the gcode. I got `Malformed command 'SET_PRESSURE_ADVANCE ADVANCE=0 EXTRUDER=ERCF 1'`

As a workaround I will remove the names from my config. I'm happy I found this issue.

GabrieleCastellani, 2022-10-16:
Hi I did in this way and it seems fixing the issue
[image]

julianjc84, 2022-12-22:
Many work around methods until the next version.
this works for me with a ERCF 9 tool
[image]

closed 2024-02-11 by supermerill