supermerill/SuperSlicer #3079: Choose angle at which a perimeter is considered an overhang?

Arakon, 2022-08-15:
**Is your feature request related to a problem? Please describe.**
On faster printers, i.e. the benchy's hull overhang is considered a normal perimeter, despite being at a 40° angle, and therefore printed at high speed, resulting in warping.

**Describe the solution you'd like**
An option to enable overhang detection at angle x, i.e to allow slower printing of that specific area.

I hope it's acceptable to just paste a screenshot here of an example I had discussed on Discord:

[image]

Pavulon87, 2022-08-16:
Perimeters & Shell → Overhags → threshold for: → Bridge speed and fan

Arakon, 2022-08-16:
Agreed, since there's a seperate Overhang Speed parameter, but basically no way to define an overhang without telling it to treat it like a bridge instead.