supermerill/SuperSlicer #3081: Export to STL with supports menu not there.

jeedee23, 2022-08-16:
### What happened?

Just mentioning that the menu export buildplate with supports to stl is not there (export buildplate exports without supports)
Ik use anycubic PWMX format so I need to export stl with supports and then Pre process with UV tools and then just print with the Anycubic Photon, which is ok

### Project file & How to reproduce

[2022-08-16(at 12h10m27s)_impeller v3.zip](https://github.com/supermerill/SuperSlicer/files/9349944/2022-08-16.at.12h10m27s._impeller.v3.zip)

### Version

2.5.59

### Operating system

win11

### Printer model

anycubic Photon MonoX  and others

neophyl, 2022-08-16:
Seems to be greyed out in PS as well.  At least as of PS v2.4.4  Wonder when they did that.

jeedee23, 2022-08-17:
Hi, My mistake, it has just changed: you can export plate and in the
following dialog just choose 'include supports'