supermerill/SuperSlicer #3085: Specify Number of Top/Bottom Walls

randellhodges, 2022-08-16:
I love that you added the "Only one perimeter" for "On first layer" and "On top surfaces". I would love to see it expanded so that instead just a single perimeter, we could choose the number of perimeters.

So if want to print with 5 walls, I could have 2 walls for the top/bottom for instance. In some cases, with infill connections, the second wall really helps and it still looks better than seeing all 5 walls.

I first alluded to it at the bottom of this issue. I don't think I've already opened a FR. It has been on the back of my mind for a while and I saw that you might be working with this code for Arachne, so I thought now would be a good time to ask.

https://github.com/supermerill/SuperSlicer/issues/1566#issuecomment-927334479

neophyl, 2022-08-16:
Just use a modifier to set how many perimeters you want.

randellhodges, 2022-08-16:
That option doesn't work correctly, which is my merill added the option for single bottom layer. If you refer to the issue I commented on, it breaks the number of bottom layers if I put a height range modifier so that my bottom layer only has 2 walls, then I only get 1 bottom layer.

neophyl, 2022-08-16:
Yes but theres usually a way to get what you want if you care to put some effort in.

Have a look at this example 
[Example.zip](https://github.com/supermerill/SuperSlicer/files/9353414/Example.zip)

2 visible perimeters on the bottom, 2 on the top and 3 in the middle.  But the number can be altered.  Yeah its not a 1 click option but the point is that if you try you can make SuSi or PS jump through hoops.