supermerill/SuperSlicer #3089: Skirt going under a few items, not around all items.

TheOddDuck, 2022-08-17:
### Version
2.3.57.12

### Operating system type + version
Windows 11 21H2 Build 22000.856

### Behavior
Skirt didn't go around all items. Instead it arbitrarily cut under a few items.
Add items - slice - the preview is accurate to how it printed
The skirt should go AROUND all items on the outside not under a few.
The skirt ended up going under some of the items.
 Screenshot attached in zip file

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
attached in a zip fil
[Clamps.zip](https://github.com/supermerill/SuperSlicer/files/9356200/Clamps.zip)
e

neophyl, 2022-08-17:
As I thought, none of your larger pieces are placed flat to the bed.  If you tried to print that plate as is, you would end up with a stringy mess.  When you have sliced get into the habit of checking the preview and scrolling through the layers.  If you had looked at just the first layer it would be obvious that the parts were at an angle.

Dont forget to add support for the over hang areas too if you are going to print this.

TheOddDuck, 2022-08-17:
@neophyl  - I did print this - a few days ago.....I had never gotten around to the bug report.

That being said - Yeah, it's a miracle it printed just fine. I completely missed that. Didn't have my mouse I normally use to zoom in :)

Sorry for the false bug report. After selecting the groups and using place on face, everything rendered properly.

closed 2022-08-17 by TheOddDuck