supermerill/SuperSlicer #3093: Weird gaps between perimeters

PuszInPusz, 2022-08-17:
### What happened?

I’m not sure if it’s some kind of compatibility problem or a weird SS bug, but after upgrading to the latest Klipper on my 2 Vorons and starting a plate for my upcoming Voron build I noticed some weird gaps in the perimeters if a couple of prints ( for now only shown on the Z drive stls and the Latch shuttle of the SB see picture) first I thought it was just unlucky and I printed those parts again and again and again even changing settings one by one didn’t help and came with the help of Ellis that it has maybe something to do with the travels ( the artifacts appear there where the nozzle travels from one point to another) Z hop is activated so don’t know what else could cause it. I run the same prints with same settings on cura to be sure that the printer isn’t the issue. Everything printed as expected and without gaps thus giving me the conclusion that the gaps are some kind of weird travel bug since I didn’t have these issues with SS before
[image]
[image]
[image]
[image]
[image]
[image]
[image]
.

### Project file & How to reproduce

[z_drive.zip](https://github.com/supermerill/SuperSlicer/files/9366139/z_drive.zip)

### Version

2.4.58.4

### Operating system

Windows 10

### Printer model

V2.4 SB and AB

CCS86, 2022-08-18:
Sort of looks like you are oozing during long travels, depositing that material where you cross a printed part, and under-extruding at the start of the next printing move, because of the lost material.

PuszInPusz, 2022-08-18:
Well… why is it happening only on those parts? Nothing else on the build plate got those weird gaps and why can’t I replicate the issue on Cura? even though running the same settings. Left SS right Cura 5.0 ( don’t Mond the seam alignment on cura) 
[image]

mashuptwice, 2023-01-27:
This might be relevant:

[image]
[image]

Both cura and prusaslicer slice the part just fine. I've tried repairing the STL with multiple services, as well as re-exported the part from the STEP file with both OnShape and Solidworks to no avail.