supermerill/SuperSlicer #3094: Arachne settings missing in 2.5.59 on Mac M1

allenrowand, 2022-08-18:
### What happened?

When I go to Print Settings/Perimeters & Shell there are no settings for Arachne. Searching using the magnifying glass returns no results
[image]
.

### Project file & How to reproduce

Steps to reproduce: launch SuperSlicer, go to Print Settings :)
I've included a project export in case there is anything in my config causing this.

[v5 v1.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9366836/v5.v1.3mf.zip)

### Version

2.5.59

### Operating system

macOS 12.5

### Printer model

Voron v2.4r2

robthide37, 2022-08-18:
is it the not arachne edition ?

allenrowand, 2022-08-18:
> is it the not arachne edition ?

That's what I thought…

salocain, 2022-08-20:
in the linux version idon't find the arachne option too.
same issue with the last nightly :-(

FedorZa, 2022-08-22:
Looks like Arachne configs disappear after using .58 stable release (configs is shared between 58 and 59, OS Linux)

salocain, 2022-08-23:
solved !! , if you don't use "defaut layout gui" ( settings-->gui ) you don't see option .

switch to "defaut layout" for seing archne param .

allenrowand, 2022-08-24:
Ok, it's definitely a config bug. I exported my config with physical printers, quit and moved the Super Slicer folder out of Application support. When I relaunched the arachne settings were there, and they persisted after I imported my config. I've attached the config that has the bug in case that helps track down the issue.

[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/9412912/SuperSlicer.zip)

Stooovie, 2022-08-24:
Same here. Two Intel Macs with Monterey, the very same Arachne version installed, both set to Expert mode, I DO see Arachne on one and DO NOT see it on the other...

supermerill, 2022-08-24:
the setting layout should be updated automatically. seems like it does not on macos...

Lunerwalker2, 2022-09-04:
I've also been experiencing this on Linux Mint (AppImage), the settings for arachne simply don't appear in 2.5.59. I know that arachne is present in the version, because I can export the config, open in a text editor, change "classic" to "arachne", and then re-import it. 

When I do that, the "Detect thin lines" becomes greyed out, indicating arachne is enabled, but there is still no GUI option for it.

nirurin, 2022-09-22:
> the setting layout should be updated automatically. seems like it does not on macos...

Also does not on windows 11 (I just came here after going through 3 of the recent versions of superslicer trying to figure out why I couldn't find the arachne toggle, when I had been told the recent nightlies were all meant to be 2.5). 

Will delete my .ini file and see if it fixes the issue.

nirurin, 2022-09-22:
Additional - At least on Windows 11

If you open the current 'RC' build (which is labelled as being the newest, and I was told is a 2.5 build, but opens up as 2.4.58), there is no arachne options in the settings (so maybe it actually is a 2.4 build?)
However if you do this, and then go and open up an arachne build, the settings will be gone. You need to delete a bunch of ini files and reopen in order to get the arrache settings back. 

So avoid the current RC build if you use arachne.

drzacek, 2022-09-25:
> Additional - At least on Windows 11
> 
> If you open the current 'RC' build (which is labelled as being the newest, and I was told is a 2.5 build, but opens up as 2.4.58), there is no arachne options in the settings (so maybe it actually is a 2.4 build?) However if you do this, and then go and open up an arachne build, the settings will be gone. You then need to delete the "ui_layout" folder to get it back.
> 
> So avoid the current RC build if you use arachne.

Hey man, big thanks for this - had same problem, downloaded the "newest" release, then found out it is without arachne, download the right one - can't find the settings. Deleting the ui_layout dir helped. 

Surely this needs to be solved on code level, cause you can't expert regular users to wonder why they lack settings that should be there.

bitingsock, 2022-09-26:
The problem seems to exist on all platforms. Windows 10 here has this problem. 2.4 ui files are **not** being overwritten by 2.5 files.

Mandragora-Arato, 2022-09-26:
It may sound like stupid question but, where do I find Nightly version? 
Best what I found is this:
https://github.com/supermerill/SuperSlicer/tree/nightly_dev
But I have no idea what to download etc. Is there some guide o something, I tried but couldn't find anything.
My OS is Win 10

nirurin, 2022-09-26:
You have to go to "actions" and then scroll downs a bit... I think there is
a link somewhere in this thread for it. I'll try and look later if you
still can't find it (I'm sending this by email I'm not on my pc atm)

Riccardo-merli94, 2022-09-29:
I fixed it by editing the version field in /Applications/SuperSlicer.app/Contents/Resources/ui_layout/default/version.ini from 
`version = 2.4.xx.x` to `version = 2.5.59.0`. Not sure if it is the best/correct way but it worked.

GKSN, 2022-10-01:
> I fixed it by editing the version field in /Applications/SuperSlicer.app/Contents/Resources/ui_layout/default/version.ini from `version = 2.4.xx.x` to `version = 2.5.59.0`. Not sure if it is the best/correct way but it worked.

Win10, same problem everything was working fine on the 5.59 tried the 2.4 and went back to the 5.59 and the arachne menu was lost.  Did the change on version.ini  and it's working :-) .

Stooovie, 2022-10-02:
> version = 2.5.59.0

Doesn't work on my Monterey Intel Mac.

salocain, 2022-10-02:
In Linux ,I need to remove the version.ini in all folder , and it's works 😁😁😁

neophyl, 2022-10-06:
Also had it happen to me on Windows 10.  I opened up someone elses 3mf file that was created in an earlier version and afterwards no arachne.  Thankfully deleting the ui folder as suggested fixed it on windows too.

user897943, 2022-10-15:
> Additional - At least on Windows 11
>
> If you open the current 'RC' build (which is labelled as being the newest, and I was told is a 2.5 build, but opens up as 2.4.58), there is no arachne options in [...]

Windows 11 here. This fixed the issue for me - deleted ui_layout on the folder:
C:\Users\<youruser>\AppData\Roaming\SuperSlicer

ncmcgrewdev, 2022-10-20:
I'm having this issue on my M1 MacBook, but it works fine on Windows 10.

ncmcgrew, 2022-10-20:
I'm having this issue on my M1 MacBook, but it works fine on Windows 10.

UnoriginalElephant, 2022-10-20:
> I fixed it by editing the version field in /Applications/SuperSlicer.app/Contents/Resources/ui_layout/default/version.ini from `version = 2.4.xx.x` to `version = 2.5.59.0`. Not sure if it is the best/correct way but it worked.

This worked for me on my M1 macbook with Monterey.

geflores5, 2022-11-10:
> > I fixed it by editing the version field in /Applications/SuperSlicer.app/Contents/Resources/ui_layout/default/version.ini from `version = 2.4.xx.x` to `version = 2.5.59.0`. Not sure if it is the best/correct way but it worked.
> 
> This worked for me on my M1 macbook with Monterey. I did have to run SuperSlicer once before making the change or else I'd get an error saying it was damaged.

Thanks! This worked on my M1 Mac Mini. Running it once before changing the .ini was key like you mentioned.

aleung, 2022-11-18:
Linux: delete the file `~/.config/SuperSlicer/ui_layout/*/version.ini`

Staubgeborener, 2022-11-21:
I can also confirm this issue with windows 10 64 bit. Arachne option will not be displayed.
As @Riccardo-merli94 wrote, you have to change the version in `version.ini` from version = 2.4.xx.x to version = 2.5.59.0.

RugerLuger, 2022-12-01:
> I can also confirm this issue with windows 10 64 bit. Arachne option will not be displayed. As @Riccardo-merli94 wrote, you have to change the version in `version.ini` from version = 2.4.xx.x to version = 2.5.59.0.

This worked for me also. Just to clarify the path to the .ini file to edit on windows 10 for me was: {wherever you extracted the .zip}\resources\ui_layout\default\version.ini

imjamessanford, 2022-12-09:
For Windows, what got Arachne to show for me is:

Help>Show Configuration Folder....then close SuperSlicer....then delete everything in that configuration folder and relaunch SuperSlicer

Arachne popped right up for me after that

BrandonGarnica, 2022-12-22:
Stumbled upon this thread because I was having the same issue on macOS (Intel - Monterey 12.4). Above was the only thing that worked for me.

> For Windows, what got Arachne to show for me is:
> 
> Help>Show Configuration Folder....then close SuperSlicer....then delete everything in that configuration folder and relaunch SuperSlicer
> 
> Arachne popped right up for me after that

You can do the same thing in macOS. Go to `Help -> Show Configuration Folder.` Closed SuperSlicer and deleted everything in the folder and relaunch SuperSlicer.  Arachne option was available after that. 

One bug that I found when Arachne is selected is that certain settings you change and reset will crash SuperSlicer. For example, when you change `Seam position` and click on the `BACK ARROW` icon or when you saved the `Seam position` and then click on the `UNLOCKED LOCK` icon, it will crash SS

der-rote-baron, 2022-12-28:
I had the same issue on Linux Mint 20.2 using the SuperSlicer-ubuntu_18.04-2.5.59.0.AppImage
Until now I used the  SuperSlicer-ubuntu_18.04-2.4.58.5.AppImage

Luckily this

> In Linux ,I need to remove the version.ini in all folder , and it's works gringringrin

did the trick for me as well (thanks [salocain](https://github.com/salocain)). Simply changing the version in the files did not work for me.

[home folder]/.config/SuperSlicer/ui_layout (or simply click "Help" --> "Show Configuration Folder")

In my case there where two sub folders, "default" and "example". Deleted the version.ini in both of them and now I have the Arachne menu in the 2.5. When I launch the 2.4 I can still see the header of the menu, but without options. Back to 2.5, menu and options are still there. So everything is fine now.

Interestingly enough, when I open the newly created version.ini's in both folders, the line for version again says 

`version = 2.4.58.3`

And yes, I launched the 2.5 first, after deleting the files

michaelmyc, 2023-04-03:
I'm on M1 Mac, and I don't see arachne settings in version 2.4.58.3, but I do see arachne settings in 2.5.59.2 after performing the aforementioned hack.

dbinenfeld, 2023-04-28:
I'm on windows 11.
I had to delete the version.ini in the default folder, after that i started the arachne version and it worked but when i started the normal version then arachne again the arachne settings were gone.
i removed version.ini again, started arachne, closed it, changet the number in version.ini to 2.5.59.0, ran normal version again and now i have a small empty box for arachne in normal version and in arachne i have the settings