supermerill/SuperSlicer #3095: IDEX Capabilities is Horrible. Ideas to make it better?

spaelectronics, 2022-08-18:
- Extruder 2 does not heat up automatically. I cannot figure this one out for the life of me.
- Tool Change Temperature does not function at all. Would like a way to have a standby temperature set on the inactive extruder.
- Some IDEX Printers use a separate part cooling fan. It must be set with the extruder active. With dual color prints, I often see one of the fans is not running.
- Start heating extruder X seconds prior to becoming active.
- Wait for inactive extruder for reach temperature before printing with it.
- Support for Duplicate & Mirror (Specifically for IDEX Printers).
- Just overall better functionality for IDEX 3D Printers would be nice. The current support is not good at all.

inspectionsbybob, 2022-09-25:
Duplicate and mirror modes on IDEX printers are a function of the printer, not the slicer.  When selecting duplicate or mirror the bed size is cut in HALF, with some extra to allow for the size of the hot end, and then the PRINTER duplicates or mirrors the commands by its self.  The only thing the slicer need to know is that the bed size has changed.  There may be some additional start gcode necessary on some printers, but on many you select D/M and just print. The slicer only sees ONE hotend and bed.
Ai am basing this comment on the JGMaker Artist that I have and use.

inspectionsbybob, 2022-09-25:
IDEX mode, multi-color or soluble supports, is another issue as there are a lot of options surrounding tool changes such as standby vs running temperatures, look-ahead pre-heating, purging and much more.  Some of these can be done in the tool change gcode, o5hers are inline, but SS is consistent with prusa/cura.
My opinion anyway

JulianGoesPro, 2023-02-11:
Hi, any update on that? Would love to get my Sigmax R19 to work with PrusaSlicer but not sure it’s worth looking into it or too much of a hassle still… thanks!