supermerill/SuperSlicer #3099: Alternating perimeter widths to improve strength

FDResearch, 2022-08-18:
Another simple sounding but probably annoying to implement feature. A setting to change the 1 perimeter extrusion width every other layer in a way to interlock them.

The idea being is each perimeter extrusion has to bond the the one next to it and so if you stack these up on a vertical wall so they are vertically aligned you make a weak point where the walls (stacks of perimeters) can delaminate from each other, if the perimeter widths are adjusted, say by changing just the external by 2x width they will stack on top of each other like a brick wall so they interlock. In practice the 2nd perimeter may be a better choice to avoid obvious artifacting.

Mostly will add strength from perimeters delaminating from each other but may also add layer bonding since each perimeter will be injecting plastic down into the slight gap, surface roughness or even bulge left between the perimeters below it adding a slight bit of surface area and keying.

Obviously this only works on vertical and mostly vertical walls but that generally how mechanical parts are designed anyway and the range of wall angles it will work is probably pretty wide, of course some algorithm to change the behavior when the wall angle makes them line up again could be made but that's way over complicated.

gswatkins, 2022-08-18:
+1 for this. 

I’ve been using extra perimeters on odd layers to achieve this, but recently looking to reduce material/weight and had the same thoughts.

supermerill, 2024-02-15:
maybe same as #2890

Sakura-sx, 2024-04-27:
> maybe same as #2890

No, that one is different, that feature creates an additional layer. CNC Kitchen has a video about it:
https://www.youtube.com/watch?v=5hGm6cubFVs
There is also an issue on PrusaSlicer:
https://github.com/prusa3d/PrusaSlicer/issues/1823

Image of vertical layer bricking:
[image]
Image of horizontal layer bricking:
[image]
Image of horizontal layer bricking without variable width adjustment:
[image]
Image of horizontal layer bricking with variable width adjustment:
[image]
*The images have are not mine. They are from the PrusaSlicer issue.

@supermerill Please implement this with the option for both layer brickings. And thank you for maintaining this cool project.

supermerill, 2024-05-03:
horizontal bricking is already implemented, see 'perimeter width on odd layers'.

vertical bricking: tell me what to do in various complicated slopes.

theboee, 2024-10-29:
@supermerill For vertical maybe have it work in conjunction with variable layer height. The slicer would attempt to fill in those overhang areas with smaller layers. The outside wall should always print normally, it would only be internal walls that are affected by bricking.

mimi89999, 2024-12-24:
Hello,
Was this feature implemented? I can't find the associated commit for this.