supermerill/SuperSlicer #3100: Arachne Ignoring "Avoid Crossing Perimeters" and fills wall gaps with solid infill instead of single perimeter line.

huntervaneck, 2022-08-18:
### What happened?

[image]
Slice with classic mode, the travel moves are very limited.
[image]
Sliced with Arachne, many more travel moves appear (significantly exacerbated when printing multiple parts at once.) Outer walls should have perimeter lines that fill the gap, not solid infill. 
[image]
Sliced again with Arachne, when the perimeter count is changed from 2 to 3 this problem changes to what I would expect from 2 perimeter walls. 3 perimeters create two outer and two inner perimeters with a thick center gap fill but also increases the number of travel moves across the center of the part.

### Project file & How to reproduce

[Faceplate.zip](https://github.com/supermerill/SuperSlicer/files/9375757/Faceplate.zip)

### Version

2.5.59.0

### Operating system

Windows 10

### Printer model

SOVOL SV01

supermerill, 2022-08-28:
With arachne, there is no gap fill. So the empty gap is given to the infill.

huntervaneck, 2022-08-28:
That would be great, because the behavior of the infill doesn’t make sense. I assumed with Arachne it would have made the two inner perimeters with a higher flow to increase theirs width and eliminate the gap. 

Any idea why the internal travel moves have almost doubled?

supermerill, 2022-08-28:
> Any idea why the internal travel moves have almost doubled?

arachne-gapfill-like-perimeter may be less flexible in their print order / orientation than gapfill.
gapfill is printed after all the periemters that enclosed them, ofte it creates unneeded travels. Arachne-gapfill-perimeters may be enforced inside the perimeter phase, and so can't be re-ordered to avoid most of the travels.