supermerill/SuperSlicer #3104: Slower perimeter speed when printing on support interface layer

D-Martijn, 2022-08-20:
**Is your feature request related to a problem? Please describe.**
When printing on the support interface layer the same default perimeter speed is used.
This causes the newly printed perimeter to sometimes come loose when printing on the interface layer.
Causing problems for the following perimeter layers to also not stick.

**Describe the solution you'd like**
Being able to have a extra setting in the speed tab to change the speed when printing perimeters on interface layers

**Describe how it would work**
see above. 

[image]

woschtkopp, 2023-04-18:
I would love to control speed above soluble support on my Idea printer, too.

woschtkopp, 2023-04-18:
Idex printer. But generally it seams like a nice setting to have. It would make fine-tuning adhesion with regular support also