supermerill/SuperSlicer #3105: Underextrusion with arachne engine

strayr, 2022-08-20:
### What happened?

Filament (polymaker polylite ABS) profile has been producing nice parts with this filament across several versions, (pressure advance is maybe a tiny bit off but that only really shows in the corners and not what I'm having trouble with here).

Printed an updated toolhead part and it came out with some gaps in the perimeters along the straight lines. 

[image]

Thinking I got my calibration wrong i printed a calibration print modified to show the perimeters and isolate the top layers from the bed with a bit of infill, this has been working more consistently fro me than the stock print. This showed great solid infill at the extrusion multiplier (0.96) I was using but with gaps in the straight sections of perimeter. 

[image]

I tried with the classic engine and got the expected results

[image]

I also tried with prusa 2.5.0RC1 and got consistent results (unmarked chips, bottom)

[image]

### Project file & How to reproduce

Open the 3mf in SuperSlicer 2.5.59.0, select a filament that needs an extrusion multiplier a little below 1, and appropriate printer. Print with arachne and classic engines.

[Arachne-flow-isues.zip](https://github.com/supermerill/SuperSlicer/files/9387585/Arachne-flow-isues.zip)

### Version

2.5.59

### Operating system

windows 10

### Printer model

stealthburner/v6 on a prusa mk2 frame and alsop with VS373

wonsuhna1, 2022-08-20:
#3043
is that the kind of problem? i'm waiting for a bug fix update.

strayr, 2022-08-20:
I think that's the same problem

closed 2024-02-11 by supermerill