supermerill/SuperSlicer #3106: Segfault

iplayfast, 2022-08-20:
### What happened?

(superslicer:507324): Gtk-WARNING **: 16:36:28.906: Failed to set text from markup due to error parsing markup: Error on line 1: Entity did not end with a semicolon; most likely you used an ampersand character without intending to start an entity — escape ampersand as &amp;
Segmentation fault (core dumped)

On layer 32 of the project I was slicing. Happens consistently.

### Project file & How to reproduce

[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/9387991/SuperSlicer.zip)
Rename the zip file to SuperSlicer.3mf 
load, and slice

### Version

Version 2.5.59

### Operating system

Linux

iplayfast, 2022-08-20:
Individual pieces can be sliced fine. Just all together.

supermerill, 2022-08-28:
same bug as #3128, fixed in next nightly

closed 2024-02-11 by supermerill