supermerill/SuperSlicer #3115: Repeatable crash slicing particular 3MF (at least on my system)

Misterke, 2022-08-22:
### Version
2.5.59.0

### Operating system type + version
Windows 10

### Behavior
* Description: Consistent crash when trying to slice attached 3mf on my system
* Reproduction: I just need to load this 3MF on my system and try to slice it
* Expected Result: No crash
* Actual Result: SuperSlicer closes without any message

[Dishwasher Brake.zip](https://github.com/supermerill/SuperSlicer/files/9397354/Dishwasher.Brake.zip)

Misterke, 2022-08-22:
I have also tried the latest nightly for Win64 and it still crashes.

neophyl, 2022-08-23:
Ok, with your printer profile selected as is in the project it also crashes for me (Win 10, v2.4.58.4).
However if I change the layer height limits Under Extruder 1 to Min 0.1 and Max 0.3 then it slices ok.    Changing minimum to 0.01 also slices fine.  I suspect something is being set to an invalid value when % is being used.  

I also tried max layer at 75% and that sliced ok too.  Minimum of 10% also works.  The weird thing is if I then change minimum to 5% again and slice it also succeeds.  But if I reload your original project which has 5% in there it crashes again.  

Thats the trouble with those, you never actually know what is being set as its calculated elsewhere.  If you have any sort of issue its best to revert to fixed values initially until you track it down :)

Definitely something weird going on but at least you can modify your profile to use fixed values for those and get slicing again.

Misterke, 2022-08-23:
> Ok, with your printer profile selected as is in the project it also crashes for me (Win 10, v2.4.58.4). However if I change the layer height limits Under Extruder 1 to Min 0.1 and Max 0.3 then it [...]

Thanks for the tip.  Indeed, modifying that does indeed seem to allow me to slice the thing.  Note that the profiles used have probably been undergone numerous updating by subsequent (alpha/beta) versions of SuperSlicer, so perhaps they contain some stuff which isn't really "visible" in the settings GUI.

supermerill, 2022-08-28:
same bug as #3128
will be fixed in next nightly.

closed 2024-02-11 by supermerill