supermerill/SuperSlicer #3118: Winget package

Zyzto, 2022-08-23:
To add superslicer to winget for easier install and update

[Winget ](https://github.com/microsoft/winget-cli) is microsoft package manager for MS_Store and msi packages that's have direct intergation with windows system to allow it to read software version even if installed manually, it can be updated through winget

both prusa slicer and cura already have winget packages.

neophyl, 2022-08-23:
Are you volunteering ?  I don't think prusa have anything to do with winget and that its been added by a volunteer.

G-Pereira, 2022-09-26:
OP is correct, there is a [PrusaSlicer package](https://www.wingetgui.com/apps?id=Prusa3D.PrusaSlicer&v=2.4.1) available with [winget](https://github.com/microsoft/winget-cli):
[image]

G-Pereira, 2022-09-26:
[Here](https://github.com/microsoft/winget-pkgs/blob/master/manifests/p/Prusa3D/PrusaSlicer/2.4.1/Prusa3D.PrusaSlicer.locale.en-US.yaml) are the package definitions

MaZe3D, 2022-09-26:
Before we can add this to winget, we need an (windows-)installer of some kind. (At least as far as I know).

G-Pereira, 2022-09-26:
That's exactly what I was thinking and a reliable host to host it. A GitHub release artifact should be fine

G-Pereira, 2022-09-26:
Also, [here](https://github.com/microsoft/winget-create#using-windows-package-manager-manifest-creator-in-a-cicd-pipeline) is how to then automate it as part of a the GitHub Actions workflow.

adripo, 2023-02-27:
Is it still possible to add this package to winget?

mindforger, 2024-07-12:
just out of curiosity and apology if i ask stupid thing, but i don't know much about windows exe and what is required for this project to run

but why do i need to install this choclatey stuff, where i need ANOTHER account for sth. that i will probably never open again 

can't we have a simple standalone .exe compiled ... i really really hate to install tools to install other tools, especially when the website of one of those tools requires me to create an account or run complicated powershell scripts .... i jsut hate more accounts to manage for things i don't really care about or install tools that might create another attackpoint on my system potentially