supermerill/SuperSlicer #3125: Arachne error - thin wall on the first layer goes to Z minus the first layer and  part of the overhang is in the air

Creo2005, 2022-08-25:
### What happened?

first layer
[image]
and overhang
[image]

### Project file & How to reproduce

[nozzle-fan-45deg.stl.zip](https://github.com/supermerill/SuperSlicer/files/9424248/nozzle-fan-45deg.stl.zip)

### Version

2.5.59.0 night last night build

### Operating system

MacOS 12.5.1

### Printer model

SapphirePlus

supermerill, 2022-08-28:
can't reproduce.
[image]

can you send the project file instead of the stl?

Creo2005, 2022-08-28:
> 

checked a minute ago
[image]

Creo2005, 2022-08-28:
[SS_config.zip](https://github.com/supermerill/SuperSlicer/files/9440079/SS_config.zip)

archive with slicer settings

Creo2005, 2022-08-28:
[screen.zip](https://github.com/supermerill/SuperSlicer/files/9440081/screen.zip)
archive with screenshots

Creo2005, 2022-08-28:
Model from this https://www.thingiverse.com/thing:3404785

supermerill, 2022-08-29:
`File`->`Save project as...` then zip it & drop it here

Creo2005, 2022-08-30:
> nozzle-fan-45deg.stl.zip

[nozzle-fan-45deg.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9449653/nozzle-fan-45deg.3mf.zip)

supermerill, 2022-08-30:
A little gcode viewer issue. The gcode file is fine.

Creo2005, 2022-08-31:
> A little gcode viewer issue. The gcode file is fine.

OK, this is good :)

closed 2024-02-11 by supermerill